snu-quiqcl/iquip #11: Simplify the subprocess and rearrange the file structure

BECATRUE, 2023-05-16:
### Feature you want to implement
I want to implement two things.
1. Simplify the subprocess
In this project, we will often use `subprocess` package to run `artiq_*` commands.
It's easy to use enough, but there are many overlapping codes.
So I suggest to create `utils.py` and implement it in this file.
The simple code looks as below:
   ```python
   import subprocess

   def run_command(cmd):
     return subprocess.run(cmd.split(" "), capture_output=True, text=True)
   ```
   Then, we can obtain the result by `run_command("blah blah").stdout`.
   See also: [How to use subprocess](https://docs.python.org/3/library/subprocess.html#using-the-subprocess-module)

2. Rearrange the file structure
In many project, there is a directory with the same name as the project and in it, there are code files.
On the outside of the directory, there are several setting and system files, such as `.gitignore`, `setup.py`, etc.

So I suggest as follow structure.
```
- iquip (root)
  |- iquip
     |- utils.py
     |- apps
        |- explorer.py
  |- .gitignore
  |- README.md
  |- ...
```

BECATRUE, 2023-05-16:
I think this should be a priority, so please leave your opinion! @kangz12345

kangz12345, 2023-05-16:
I agree with the file structure, thank you for suggesting!

However, I suggest using a different name for the command utility module.
The name `utils` seems too general, so it could be confusing later: _What kind of utility functions are here?_

For example, when we need more _utility functions_, we have to put them in the same module even if they are totally not related to each other.
Moreover, we are going to make `protocols` module, so the name `utils` would become more ambiguous.

In my opinion, more specific names like `command`, `cmdtools`, etc. would be better, depending on the actual contents of the module!

BECATRUE, 2023-05-17:
I agree that the name `utils` may cause a little confusion.
Then, I will choose the name `cmdtools`!

BECATRUE, 2023-05-23:
This issue is completed in #13, so I closed it.

closed 2023-05-23 by BECATRUE