snu-quiqcl/iquip #15: Single ARTIQ TTL monitor widget

kangz12345, 2023-05-19:
### Feature you want to implement
A widget and object which monitors an ARTIQ TTL input or output.
When the input or output changes, the monitor should be able to notice it.
Note that this does not include any `qiwi` app.
It will be implemented in the next issue.

### How the feature is implemented
Create a module `monitor.py` and implement classes `TTLMonitor` and `TTLMonitorWidget`.

#### `TTLMonitor`
- It is a non-GUI object which monitors a single TTL channel.
- Specific implementation will be decided later.

#### `TTLMonitorWidget`
- It is a PyQt widget for displaying a single TTL channel status.
- It does not include the proper slot connections to actually update the status. It only includes the GUI part, without the core logic.

BECATRUE, 2023-05-22:
> It will be implemented in the next issue.

What will be implemented in the next issue?

BECATRUE, 2023-05-22:
In according to your plan, is the `TTLMonitor` a subclass of `BaseApp` and is the `TTLMonitorWidget` a frame of the former class?

kangz12345, 2023-05-22:
> What will be implemented in the next issue?

The `qiwis` app and frames will be implemented later.

> In according to your plan, is the `TTLMonitor` a subclass of `BaseApp` and is the `TTLMonitorWidget` a frame of the former class?

No, the classes implemented in this issue are independent to `qiwis`.
They will be used to implement `qiwis` apps and frames.

BECATRUE, 2023-05-23:
Now I understand! Thanks for your explanation.

PR #23 (2023-05-28): Single TTL monitor widget
This closes #15.

In this PR, only the `Monitor` and `TTLMonitorWidget` are implemented.
Since they are too _abstract_, I will make unit tests for those after this PR.
closed 2023-05-28 by kangz12345