snu-quiqcl/iquip #17: Asyncio support

kangz12345, 2023-05-21:
## Problem
Since `artiq` is using `asyncio` for their socket communications, we have to support `asyncio` if we want to import it and use it in the codes.
As far as I understand, we cannot use `asyncio` and `PyQt5` at the same time because both of them require a event loop in the main thread.
Therefore, we have to use `qasync.QEventLoop` just like the ARTIQ dashboard does.

However, since we use `qiwis`, we have to modify `qiwis` to support `qasync`, which seems not so simple.
Moreover, I don't have much experience in `asyncio`.
If we study more about `asyncio`, then it might not be a big problem.

### Why do we need to import `artiq`, besides running command-line scripts?
For instance, to implement artiq device monitors, we need [CommMonInj](https://github.com/m-labs/artiq/blob/df15f53ee97558b31c23a3bbe93f32c14b18fe74/artiq/coredevice/comm_moninj.py) and it uses `asyncio`.

## Alternative
One of the simplest alternative would be to implement `CommMonInj` without `asyncio` by ourselves.
The downside of this approach is that we have to manually update the codes whenever the `artiq` library codes or device protocols are updated.
Therefore, this approach is not scalable and is very unstable.

## We have to decide
1. Study more about `asyncio` and support it in `qiwis`.
2. Take the alternative with the downsides.
3. Make a pull request `artiq` to separate the socket reader and writer from `CommMonInj`.

BECATRUE, 2023-05-22:
Wow.. That's a keen observation, but it's too bad there is no neat solution.
I think the first solution is better that the second one because the second is unstable and difficult to check its safety as you said.

In the past, I thought that it would be better to simply create an experiment that measures the value of the devices and monitor the result periodically.
But this seems too inefficient, doesn't it?

kangz12345, 2023-05-23:
I agree with you that we should go with the first option.

However, since we have to study more and determine if it could be a feasible solution, for now your suggestion about the monitor seems acceptable.

In fact, our device communication protocol was _user-driven_, i.e., devices send messages to users only when the users query a question.
In other words, devices never send messages by itself.
Therefore, I think we will not feel any inconvenience even if we don't have the _realtime_ monitor feature.

However, disabling the existing feature is definitely not our goal. We should implement the feature later.

BECATRUE, 2023-05-23:
I see!

Then, let's implement it as running an experiment which fetch the device status and make a note about what to do later.

kangz12345, 2023-05-23:
We need to study more about `asyncio`:
1. Is `asyncio` stable and modular enough for `qiwis` to support? Is it feasible for the apps to use `asyncio` freely?
2. Is it possible to run `asyncio` with different event loop, so that the `asyncio` never interleaves the `PyQt` event loop.

BECATRUE, 2023-06-02:
It is related to #31.

BECATRUE, 2023-06-16:
I think this issue is also unnecessary thanks to the proxy server.
Then, can we close this issue? @kangz12345

kangz12345, 2023-06-16:
I agree with you. 👍

closed 2023-06-16 by kangz12345