snu-quiqcl/iquip #54: Upgrade the Python version

kangz12345, 2023-07-05:
As [Status of Python Versions](https://devguide.python.org/versions/) says, Python 3.7 became _end-of-life_ since 2023-06-27.
We use Python 3.7 in the current experiment PC, so I think we should upgrade it to the latest version Python 3.11.
To do so, we may upgrade it step by step as introduced in [an article](https://pythonspeed.com/articles/stop-using-python-3.7/):
>
1. Upgrade to 3.8.
2. Fix any bugs you find.
3. Upgrade to 3.9, fix any bugs.
4. Repeat until you hit Python Python 3.11.

### Original comment
To obtain the raw code text from the ast tree (column 2 of the viewer), I used [`ast.get_source_segment()`](https://docs.python.org/3/library/ast.html#ast.get_source_segment).
However, this is only supported after version 3.8, and our supporting lowest version is 3.6 so it makes GitHub action errors.

As @kangz12345 and I said briefly that the version 3.6 is too old, how about raise the lowest version to 3.8 or 3.9?

_Originally posted by @BECATRUE in https://github.com/snu-quiqcl/iquip/issues/53#issuecomment-1620932898_

BECATRUE, 2023-07-06:
It looks cool! Then, how about starting this issue after merging up to #47?

kangz12345, 2023-07-06:
Do you mean after merging #47?

BECATRUE, 2023-07-06:
Yes! Do you have another plan?

kangz12345, 2023-07-07:
😄 No I just wanted to check the meaning of "up to".

kangz12345, 2023-07-20:
Oh we forgot this after merging #47. I will update the GitHub actions.

PR #73 (2023-07-20): Remove Python 3.7 and add 3.11 to GitHub actions
This closes #54.
commit 3cc24256 (2023-07-20): Remove Python 3.7 and add 3.11 to GitHub actions (#73)

This closes #54.
closed 2023-07-20 by kangz12345