snu-quiqcl/iquip #55: Handle an empty `choices` of enumeration value

BECATRUE, 2023-07-06:
Oh, now I understand. Okay then, please create an issue for handling an empty enumeration value.

_Originally posted by @kangz12345 in https://github.com/snu-quiqcl/iquip/pull/47#discussion_r1252489188_

BECATRUE, 2023-08-04:
I think it is enough to leave a `WARNING` or `ERROR` log because it is not a problem in terms of IQUIP, but just the experiment code.
What do you think of it? @kangz12345

kangz12345, 2023-08-04:
Yes, I think it should leave an `ERROR` log and not open the builder window (or, if it should be checked in the builder app, terminate the builder app).

BECATRUE, 2023-08-07:
How about just showing a warning alert?
I think it is enough because even after opening the builder, the user can fix the experiment code and apply it to GUI by clicking the reload button.

In addition, let's imagine the following situation.
1. Open a builder.
2. Update the experiment code to make `choices` empty.
3. Reload the build arguments.

In that case, terminating the builder app looks a little awkward.

Thus, I think alerting is enough.

@kangz12345

kangz12345, 2023-08-07:
Okay then does it alert when the user tries to submit the experiment (when the submit button is clicked)?

BECATRUE, 2023-08-08:
Originally I planned that it alerts when the builder app is created or reloaded, but you suggestion would be better!

PR #99 (2023-08-10): Handle an empty `choice`s of enumeration value
This PR will close #55.

To handle an empty `choice` of enumeration value, I implemented the following:
1. When an `_EnumerationEntry` has an empty `choice` and its `value()` is called, it raises a `ValueError`.
2. Once the submission button is cliced, the app obtains each value of all entries and if a `ValueError` occurs, the submission is rejected.

### Result
[image]

In this case, the `color` argument has an empty `choice` and if the submission button is clicked, several logs are printed.
commit a1b3f7d1 (2023-08-10): Handle an empty `choice`s of enumeration value (#99)

This PR will close #55.

To handle an empty `choice` of enumeration value, I implemented the
following:
1. When an `_EnumerationEntry` has an empty `choice` and its `value()`
is called, it raises a `ValueError`.
2. Once the submission button is cliced, the app obtains each value of
all entries and if a `ValueError` occurs, the submission is rejected.

### Result

[image]

In this case, the `color` argument has an empty `choice` and if the
submission button is clicked, several logs are printed.
closed 2023-08-10 by BECATRUE