snu-quiqcl/iquip #63: listing pending experiments and displaying currently running experiment in `schedule.py`

giwon2004, 2023-07-07:
### Feature you want to implement
Implement a scheduler that shows the list of experiments and their status

### How the feature is implemented
In `iquip/apps`, make a new file named `schedule.py` to implement the GUI, using `qiwis`, according to the python code [artiq/artiq/dashboard/schedule.py](https://github.com/m-labs/artiq/blob/master/artiq/dashboard/schedule.py).

`schedule.py` includes a class named `SchedulerFrame` which has `QListView` for listing pending experiments. (Class `ScheduleDock` in [artiq/artiq/dashboard/schedule.py](https://github.com/m-labs/artiq/blob/master/artiq/dashboard/schedule.py) used `QTableView` to show information like `expid` and `status` in a table, but I want to customize the items inside the list.)
 + Here, I want to take advantage of `QListView.Movement = QListView.Free` to enable changing the priority of experiments in the scheduler GUI. But if this does not work, then I would try hardcoding the drag & drop property later.

An isolated frame (for displaying the current experiment) and this `QListView` will contain instances of the class `ExperimentView`, which will be later implemented.

BECATRUE, 2023-07-16:
> Here, I want to take advantage of QListView.Movement = QListView.Free to enable changing the priority of experiments in the scheduler GUI. But if this does not work, then I would try hardcoding the drag & drop property later.

It looks so cool property!
However, I'm worried that changing the priority looks difficult, and I think you should cancel the experiment and re-sumbit it.
If you think it would become a big implementation, you can separate into several issues!

giwon2004, 2023-07-16:
Oh, I did not know that changing priorities was a big deal.

If it is, I guess I can implement everything except this drag & drop feature first, and then try it.

PR #69 (2023-08-02): Giwon2004/63/scheduler
The `scheduler.py` is implemented in a _local_ sense, which does not interact with `artiq-proxy` in a moment, and will be implemented afterward. 

The list `expList` is implemented through `QListView`, which holds `ExperimentView` instances.
The model `ExperimentView` inside `expList` also provides a drag & drop feature.

### Examples
+ Here, I had no connection through `artiq-proxy` for now, so I simply added some dummy experiments before starting.

[image]
+ Current experiment is at the top, while others are on the list.
   - Other information besides their names is di [... cut]
closed 2023-08-02 by giwon2004