snu-quiqcl/iquip #64: implement `ExperimentView` for displaying experiment information

giwon2004, 2023-07-07:
### Feature you want to implement
Implement a class `ExperimentView` that shows the experiment information

### How the feature is implemented
`ExperimentView` displays the information of the corresponding experiment (the class that holds this information will be a replica of the `Model` class in [artiq/artiq/dashboard/schedule.py](https://github.com/m-labs/artiq/blob/master/artiq/dashboard/schedule.py).) that has some interactions, such as showing detailed information, deleting, changing priorities, changing scannable values (this would need another UI, which will be implemented later, mentioned in [Discussion#62](https://github.com/snu-quiqcl/iquip/discussions/62#discussion-5377377)), and et cetera.

+ It needs to enable the selection of experiments by left click, and showing the menu when right-clicked but I am not sure about the compatibility between `QListView.Movement = QListView.Free` and this click event-based feature.

BECATRUE, 2023-07-16:
Then, did you plan that each item of `QListView` is an `ExperimentView` instance?

giwon2004, 2023-07-16:
Yes, actually, no.

`QListView` takes `QAbstractListModel` (where drag & drop is also implemented) as a model for the data structure.
So `ExperimentModel` inherits `QAbstractListModel` to manage data - which is the list of `ExperimentView` instances - of `QListView`, with the assist of `ExperimentDelegate` which _displays_ the layout implemented in `ExperimentView` inside `QListView`.

For more information, please check this out: [giwon2004/63/scheduler/iquip/apps/scheduler.py#L39](https://github.com/snu-quiqcl/iquip/blob/giwon2004/63/scheduler/iquip/apps/scheduler.py#L39)

kangz12345, 2023-07-17:
The `style` label is for _code_ style.

giwon2004, 2023-07-17:
Sorry, I thought making `ExperimentView` rather than manually adding elements is a type of styling the code.

kangz12345, 2023-07-17:
No problem!

kangz12345, 2023-07-17:
> No problem!

Oh, I misunderstood your reply. I will explain more about it.
We usually use `style` label when we update the existing codes to adapt a new style.

This issue is for a new feature and not dealing with the current code style, so it is out of the scope of `style` label.

giwon2004, 2023-07-19:
Oh, I understood. I will manage the tags that way from now. Thank you!

closed 2023-08-02 by giwon2004