snu-quiqcl/iquip #81: Explorer: click open button without selecting any item

kangz12345, 2023-07-27:
### Describe the bug
When you click the open button without selecting any item (the initial state), the program aborts.
Or the reason is that the file tree is empty (when I cannot access the artiq-proxy or the file tree is actually empty), I did not check which is the reason.

### Expected behavior
Open button is disabled or nothing happens even if I click the open button.

### To Reproduce
Steps to reproduce the behavior:
1. I tested this in [giwon2004/63/scheduler](https://github.com/snu-quiqcl/iquip/tree/giwon2004/63/scheduler) branch.
2. Run `iquip` by console command `qiwis` in the repository.
3. No access to `artiq-proxy` server.
4. The file tree is empty
5. Click the open button.

### Expected causes and correction methods
Check if `experimentFileItem` is `None` in `openExperiment()`.
Or do some more signal-slot stuffs to disable the open button when no file is selected.

### Screenshots
[image]

BECATRUE, 2023-07-27:
I reproduced this situation, and I think both reasons you suggested look correct😂

kangz12345, 2023-07-27:
Thank you for the investigation! Then the first reason (not selected) is correct (because when the file tree is empty, we cannot select any item).

PR #95 (2023-08-06): BUG: Click open button of explorer without selecting any item
This PR will close #81.

I resolved the bug that without selecting any item, `open_experiment()` causes an error.
commit 7acd7c6f (2023-08-06): BUG: Click open button of explorer without selecting any item (#95)

This PR will close #81.

I resolved the bug that without selecting any item, `open_experiment()`
causes an error.
closed 2023-08-06 by BECATRUE