snu-quiqcl/iquip #93: Implement an explorer for visualization

BECATRUE, 2023-08-03:
### Feature you want to implement
We need an explorer to show the visualize path through the proxy server.
For this, I will implement `VisualizeExplorerApp`, similar to `ExplorerApp`.

In this app, there are the following functions.
- After receiving the visualize path structure, show each submission info in the explorer.
- Add a button to reload the structure.
- Add a button to open a specific visualized result.

### How the feature is implemented
According to snu-quiqcl/artiq-proxy#36, in the visualize path, there are only directories with a RID name, not files.
(It means that the structure is flat.)
Thus, we can show the structure using `QListWidget`, not `QTreeWidget`.

However, I want to use `QTreeWidget` because under each RID directory, I will add the submission information, i.e., build argumetns, priority, due date, etc. as a child `QTreeWidgetItem` of the directory.

### Additional context
After implementing to fetch the submission information in snu-quiqcl/artiq-proxy#42 and snu-quiqcl/artiq-proxy#38, I will start to develop this.

kangz12345, 2023-08-04:
How about "ResultExplorer" instead of "VisualizeExplorer"?
I think we can add more features for it, e.g., link to data viewer, etc.
The essence of it is, in my opinion, exploring the experiment results.

BECATRUE, 2023-08-07:
> How about "ResultExplorer" instead of "VisualizeExplorer"? I think we can add more features for it, e.g., link to data viewer, etc. The essence of it is, in my opinion, exploring the experiment results.

@kangz12345 
I hope to find out more appropriate name of this app, too!
However, "ResultExplorer" looks different from "VisualizeExplorer" because even if the experiment was not visaulized, the result is always generated.

BECATRUE, 2023-08-07:
However, I still hope to find out more appropriate name of this app, too!

kangz12345, 2023-08-08:
Oh.. I thought the h5 files will also be saved in those directories. Then currently only vcd files are included?
If we want to explore the h5 files, should the data viewer app implement another explorer?

BECATRUE, 2023-08-24:
As @kangz12345 and I discussed, I updated this plan!

PR #120 (2023-09-18): Implement H5 result viewer
This will close #93.

# What was implemented
- H5 result viewer in `resultInfoTree`.
- The implementation is similar to the explorer app.

# Results
Before clicking a RID

[image]

After clicking a RID

[image]
commit b0e9bcb1 (2023-09-18): Implement H5 result viewer (#120)

This will close #93.

# What was implemented
- H5 result viewer in `resultInfoTree`.
- The implementation is similar to the explorer app.

# Results
Before clicking a RID

<img width="60%"
src="https://github.com/snu-quiqcl/iquip/assets/76851886/b04dab12-b5e0-4040-b55e-8f131152788d">

After clicking a RID

<img width="60%"
src="https://github.com/snu-quiqcl/iquip/assets/76851886/a5ae3e6a-89d2-4892-a314-d4aa178c79e9">
closed 2023-09-18 by BECATRUE