snu-quiqcl/iquip #106: Set another viewer for deleted/terminated/completed experiments.

giwon2004, 2023-08-13:
### Feature you want to implement
I want to set another viewer to list experiments that are not on the scheduler now but were on the scheduler.

### How the feature is implemented
The deleted, terminated experiments will now be sent to another `QListView` that holds them.
When the experiment is gone after its `status` being `"deleting"`, that experiment will be also listed in that `QListView` instance.

Right-clicking them will show another `QMenu` with one element: `Restart experiment`, submitting the same experiment again.
This feature would use the [code in artiq-proxy](https://github.com/snu-quiqcl/artiq-proxy/blob/develop/main.py#L148) that was not originally meant to be used by `scheduler.py`.

kangz12345, 2023-08-16:
Great. Will it be another app, or just another frame in the same app (`SchedulerApp`)?

giwon2004, 2023-08-16:
For now, I think it will be inside the same app, with two parallel lists.

kangz12345, 2023-11-21:
As the previous GUI is deprecated, I close this issue.
However, the feature would be necessary and might be implemented later.

closed 2023-11-21 by kangz12345