snu-quiqcl/iquip #116: Add more methods in `SubmittedExperimentInfo`

giwon2004, 2023-09-01:
### Feature you want to implement
There are more uses of `SubmittedExperimentInfo` in [`protocols.py`](https://github.com/snu-quiqcl/iquip/blob/develop/iquip/protocols.py) (mentioned in the [discussion](https://github.com/snu-quiqcl/iquip/pull/91#discussion_r1312673755) in #91), so I want to make new methods of `SubmittedExperimentInfo` here.

### How the feature is implemented
1. Add `keys()` to have lists of attributes.
2. Add `__lt__()` and other comparing functions to set orders by `priority` value.
3. Add `__repr__()` to use in debugging.

### Additional context
I want this branch to be clear, so I will make a branch after #113 is merged.
And as this is from #91, this will contain modifications in `scheduler.py` too, so #91 must be also merged before making the branch.

kangz12345, 2023-09-01:
Actually, I think using the `dataclasses`-native API would be better.
For example, you can use [`dataclasses.fields()`](https://docs.python.org/3/library/dataclasses.html#dataclasses.fields) instead of defining `keys()`.
Moreover, if you see the arguments for [`dataclasses.dataclass()`](https://docs.python.org/3/library/dataclasses.html#dataclasses.dataclass) decorator, there are `order` and `repr`.
If you set `order=True`, `__lt__()` will be implemented as the document says, then you can put the `priority` field at the top among the other fields, to make it have the highest comparison priority.

giwon2004, 2023-09-02:
Then this issue will have no significance 😂

I will close this issue.

giwon2004, 2023-09-02:
Oh, adding `order = True` is quite a change to the class. I will make a small PR here.

PR #118 (2023-09-04): Add `order` option to `SubmittedExperimentInfo`
This simple addition sets the instances' key value to sort by its defined order; the `priority` value, `rid` as its tiebreaker, `status` as its tiebreaker's tiebreaker, and so on.

```
@dataclasses.dataclass(order = True)
class SubmittedExperimentInfo:
    priority: int
    rid: int
    status: str = ""
    pipeline: str = ""
    expid: Dict[str, Any] = dataclasses.field(default_factory = dict)
    due_date: str = ""
```

This closes #116.
commit 3e6f2ec5 (2023-09-04): Add `order` option to `SubmittedExperimentInfo` (#118)

This simple addition sets the instances' key value to sort by its
defined order; the `priority` value, `rid` as its tiebreaker, `status`
as its tiebreaker's tiebreaker, and so on.

```
@dataclasses.dataclass(order = True)
class SubmittedExperimentInfo:
    priority: int
    rid: int
    status: str = ""
    pipeline: str = ""
    expid: Dict[str, Any] = dataclasses.field(default_factory = dict)
    due_date: str = ""
```

This closes #116.
closed 2023-09-04 by giwon2004