snu-quiqcl/iquip #124: Make unit test success

giwon2004, 2023-09-12:
From PR #91, we encountered a problem with the' _ExperimentQueueFetcherThread' instance running through the entire app.

This is forcibly destroyed when each unit test ends, and this makes an error.

It needs a fix and is separated from #91 to handle this only.

kangz12345, 2023-09-13:
Did you find any clue? I'm sorry that I didn't have much time to think about this.

giwon2004, 2023-09-13:
@BECATRUE and I are trying to fix this. I will try to find more information about the solution.

kangz12345, 2023-09-13:
Great! Since this makes the unit test abort, it should be fixed as soon as possible.
However, we need some necessary features for our experiment, I decided to develop further even if this issue is not resolved soon.
After implementing the mandatory features, I will take a deeper look in here.
Of course whenever you need some help, please ask me!

BECATRUE, 2023-09-13:
Wow, I realized that there is no tests about threads in the unittest, so we MUST mock all threads.
If so, all errors get rid off!

Sure, we need to implement tests about threads, but now we can resolve this issue through that.

PR #135 (2023-09-14): Fix the bug of unittest in SchedulerApp
This closes #124.

To prevent the threads killed, I mocked all threads.
commit fbc89857 (2023-09-14): Fix the bug of unittest in SchedulerApp (#135)

This closes #124.

To prevent the threads killed, I mocked all threads.
closed 2023-09-14 by BECATRUE