snu-quiqcl/iquip #132: [BUG] Several errors when fetching the scheduled queue

BECATRUE, 2023-09-13:
### Describe the bug
There are two main problems:
- The scheduler app is fetching the queue so slowly. Even though I submit several experiments, there is no update.
I didn't look up this deeply, so it may be a problem of ARTIQ-PROXY, not here.
- Once fetching the queue, the following error occurs.
[image]

kangz12345, 2023-09-13:
Did you apply the change #129?

BECATRUE, 2023-09-14:
> Did you apply the change https://github.com/snu-quiqcl/iquip/pull/129?

I will check it!

Also, the first bug seems to be referred to #136 by @giwon2004.

BECATRUE, 2023-09-14:
I pulled the `develop` branch a few minutes ago, but the same error occurs.

BECATRUE, 2023-10-14:
I renewed the scheduler app in #181, so I close this issue.

closed 2023-10-14 by BECATRUE