snu-quiqcl/iquip #136: Queue is not fetched

giwon2004, 2023-09-13:
### Describe the bug
We have this code in the [`scheduler.py`](https://github.com/snu-quiqcl/iquip/blob/develop/iquip/apps/scheduler.py#L307):
```
         response = requests.get(
                    f"http://{self.ip}:{self.port}/experiment/queue/", timeout=10
                )
```
but this continues to give `Timeout` error, while [`main.py`](https://github.com/snu-quiqcl/artiq-proxy/blob/develop/main.py#L107) has the function:
```
@app.get("/experiment/queue/")
async def get_experiment_queue() -> Dict[int, Dict[str, Any]]:
    """Gets the list of queued experiment and returns it.

    Returns:
        A dictionary of queued experiments with rid as their keys.
        The value of each corresponding rid is a dictionary with several information:
          "priority", "status", "due_date", "pipeline", "expid", etc.
        It includes the running experiment with different "status" value.
    """
    remote = get_client("master_schedule")
    current_queue = copy.deepcopy(previous_queue)
    while current_queue == previous_queue:
        await asyncio.sleep(0)
        current_queue.clear()
        current_queue.update(remote.get_status())
    previous_queue.clear()
    previous_queue.update(current_queue)
    return current_queue
```

### Expected behavior
It must return when the queue is modified.
Thus, when an experiment is submitted, then `experiment/queue/` must be called.

### To Reproduce
Steps to reproduce the behavior:
1. With the `develop` branch, open a simulation and submit it.
2. Nothing shows up.

### Expected causes and correction methods
It is sure that `Timeout` error occurs, and I guess it is caused by something that halts `main.py` from running the thread.

### Screenshots
[image]

kangz12345, 2023-09-14:
Tip: when you write a long code using "```", you can use syntax highlighting by writing "```python", as I edited your comment.

BECATRUE, 2023-10-14:
Since I renewed the scheduler app in #181, I close this.

closed 2023-10-14 by BECATRUE