snu-quiqcl/iquip #143: `DataViewerApp`

kangz12345, 2023-09-15:
### Feature you want to implement
`DataViewerApp` features:
1. Source selection: Choose the data source among the current experiment (realtime), the result in the artiq-master (remote), the local file (local).
2. Histogram plot and state discrimination threshold & data type (P1 population, average photon count, total photon count) configuration
3. Main plot.
4. Tools, e.g., sinusoidal fitting
[image]

Editing...

### How the feature is implemented
ex) Add *B* method to *A* class, Upgrade *C* model

### Additional context
Add any other context or screenshots about the feature here.

kangz12345, 2023-09-15:
> ### Feature you want to implement
> `DataViewerApp` features:
> 
> 1. Source selection: Choose the data source among the current experiment (realtime), the result in the artiq-master (remote), the local file (local).
> 2. Histogram plot and state discrimination threshold & data type (P1 population, average photon count, total photon count) configuration
> 3. Main plot.
> 4. Tools, e.g., sinusoidal fitting
>    Conceptual drawing (sorry for the bad quality ;)):
> 
> [image]
> ### How the feature is implemented
> * First, a minimal version will be implemented: #144

BECATRUE, 2023-09-16:
I'm little confused about the difference between "remote" and "local".
As I understood, "remote" method is fetching the H5 file from the proxy server and showing the data like `VisualizerApp`.
Is it right?

kangz12345, 2023-09-16:
Yes, correct! "remote" shows the result data from the artiq master, so it requires the RID of the desired experiment.
On the other hand, "local" is for showing data files in local (machine). This could allow the data viewer to be more versatile. It don't need the artiq master.