snu-quiqcl/iquip #144: `DataViewerApp` with minimal features

kangz12345, 2023-09-15:
### Feature you want to implement
What is the feature you want to implement?

### How the feature is implemented
ex) Add *B* method to *A* class, Upgrade *C* model

### Additional context
Add any other context or screenshots about the feature here.

kangz12345, 2023-09-18:
Oh, since I created a branch from this issue, #146 automatically closed this issue.

BECATRUE, 2023-09-19:
I think the example plot (in your hand painting) will be shown when the P1 radio button is selected.
Then, what plot will be shown when the other radio button is selected?

kangz12345, 2023-09-19:
The radio button does not change the plot here. Instead, it will change the main plot.
The histogram is helpful regardless of the main plot type (total/average/p1).

BECATRUE, 2023-09-19:
Oh, I understand it now! Thanks😃

PR #176 (2023-09-26): Minimal `DataViewerApp` part 4: Final
This closes #144.

I'm sorry for the large diffs. I will try to separate it if you feel too much.

Initial state:
[image]

After clicking "Sync":
[image]

## Dataset user guide
Users have to build their dataset in the following format:
```python
class Test(EnvExperiment):
    def prepare(self):
        self.set_dataset("name", [], broadcast=True)  # must not be in @kernel
        self.set_dataset("name.parameters", ["Param1", "Param2"], broadcast=True)
        self.set_dataset("name.units", ["Unit1", ""], broadcast=True)  # no unit for param2

    @kernel
 [... cut]
closed 2023-09-26 by kangz12345