snu-quiqcl/iquip #145: Pass the unittest without any error

BECATRUE, 2023-09-15:
### Feature you want to implement
After updating QIWIS version, it causes too many unittest error due to global constants.
Thus, I will resolve this temporarily, not completely. (Just make the unittest passed)

kangz12345, 2023-09-15:
If it is a temporal fix, please do not close this issue until we resolve this completely.

BECATRUE, 2023-09-15:
Could you check this branch?

First, I fixed only `test_explorer.py` to use global constants!
If you think it is okay, I will apply the same method to other apps.

kangz12345, 2023-09-15:
I think instead of assigning the constants namespace like
```python
explorer.ExplorerApp._constants = CONSTANTS
```
mocking it in `setUp()` would be better.
```python
patcher = mock.patch("explorer.ExplorerApp._constants", CONSTANTS)
```

BECATRUE, 2023-09-15:
Okay! Then, I will apply your suggestion.