snu-quiqcl/iquip #152: `DeviceMonitorApp`

BECATRUE, 2023-09-19:
### Feature you want to implement
To monitor and control several ARTIQ hardwares, e.g., TTL, DDS, DAC, I will develop `DeviceMonitorApp` in the following issues.

__NOTE__: As discussed with @kangz12345, for the first step, I will implement only a simple TTL controller, not showing the value.
This will be updated as soon as it materializes.

#### TTL controller
In `DeviceMonitorApp`, `TTLControllerFrame` will be implemented, which monitors and controls __ALL__ TTLs described in `config.json`.
This frame consists of 1. a group of `TTLControllerWidget` which monitors and control __A_ TTL and 2. a button for overriding.
The `TTLControllerWidget` has an instance of `TTLMonitorWidget` in "iquip/monitor.py" and several widgets, i.e., a level button.

### How the feature is implemented
#### TTL controller
1. Single TTL controller: `TTLControllerWidget`
2. Frame with all TTL controllers: `TTLControllerFrame`
3. Add functions to run actually: `DeviceMonitorApp`

BECATRUE, 2023-09-19:
This is the first version, so I planned to distinguish the states of overriding and level through only the status of its toggle button, not changing background color or border width of each TTL widget.

Is it okay? @kangz12345

kangz12345, 2023-09-19:
No problem, checkable buttons must be enough (`QPushButton` with `setCheckable(True)`, for example).

BECATRUE, 2023-10-16:
All sub-issues are completed😃

closed 2023-10-16 by BECATRUE