snu-quiqcl/iquip #181: New `SchedulerApp`

BECATRUE, 2023-10-04:
### Feature you want to implement
I will develop a new version of scheduler app because of the following reasons:
- `QAbstractTableModel` looks more useful than `QAbstractListModel` since the scheduled experiment list is a 2D dataset;
a row is each experiment, and a column is each info in an experiment.
- In current scheduler app, there are several bugs.

### How the feature is implemented
Basically, it will be implemented under the [model/view architecture](https://doc.qt.io/qtforpython-5/overviews/model-view-programming.html).
If you're not familiar to this, I strongly recommend to read it. (I'm also unfamilar to it..😅)

I briefly separate this issue into four minor issues:
1. Construct the model/view architecture in order to show the given scheduled queue.
2. Fetch the actual queue from the proxy server, and apply it actively.
3. Enable to abort and delete a specific experiment.
4. Enable to modify a specific queued experiment.

BECATRUE, 2023-10-10:
@kangz12345 
Is the final issue; modifying a specific queued experiment, necessary?
I planned that issue, but I don't know well if it is really necessary.

kangz12345, 2023-10-11:
I don't think it is necessary for now. You can postpone that issue.

PR #197 (2023-10-13): Scheduler: Enable to delete and terminate a specific experiment
This closes #192, and finally closes #181.

Now, you can delete a specific experiment by right-clicking.

### Screenshot
[image]
commit 2c51e58d (2023-10-13): Scheduler: Enable to delete and terminate a specific experiment (#197)

This closes #192, and finally closes #181.

Now, you can delete a specific experiment by right-clicking.

### Screenshot

[image]
closed 2023-10-13 by BECATRUE