snu-quiqcl/iquip #201: Apply QIWIS v2.3.0

BECATRUE, 2023-10-16:
### Feature you want to implement
Apply the newest version of QIWIS.

kangz12345, 2023-10-20:
Since we don't have to change any implementation for `qiwis@v2.3.0`, how about applying the change when it is necessary for the later version?
The only thing we can do is changing the GitHub actions, but it does not make any practical difference.
I guess users can just upgrade `qiwis` locally and simply use it.

BECATRUE, 2023-10-20:
I see! The next version will include setting a frame title, so after done, I will apply the newest version.

kangz12345, 2023-11-21:
@BECATRUE Could you apply this update?

BECATRUE, 2023-11-21:
Oh, I planned to update it until tomorrow.
Thanks for notifying me.

BECATRUE, 2023-11-22:
In QIWIS, I found a bug in snu-quiqcl/qiwis#243, hence I will release the modifed version soon.

This issue is compatible with the updated version, not v3.0.0.
(It works without any error, but shown frame titles are wrong.)

PR #222 (2023-11-22): Apply QIWIS v3.0.1
Please review this after snu-quiqcl/qiwis#244 is merged.
Before the newest QIWIS version is released, all actions will be failed due to non-existing QIWIS version.

### Screenshot
[image]

This closes #201.
commit 8e379749 (2023-11-22): Apply QIWIS v3.0.1 (#222)

Please review this after snu-quiqcl/qiwis#244 is merged.
Before the newest QIWIS version is released, all actions will be failed
due to non-existing QIWIS version.

### Screenshot

[image]

This closes #201.
closed 2023-11-22 by BECATRUE