snu-quiqcl/iquip #203: `selectDataPoint` fails when a dataset is loaded

BECATRUE, 2023-10-16:
### Describe the bug
When a new dataset is loaded, an exception occurs.

[image]

### To Reproduce
Steps to reproduce the behavior:
1. Type a dataset name.
2. Turn on polling for a while.
3. Set an appropriate axis.
4. Click "Apply" button.

### Expected causes and correction methods
As I debugged, [DataViewerApp.dataPoint()](https://github.com/snu-quiqcl/iquip/blob/develop/iquip/apps/dataviewer.py#L984) doesn't support `index` with `(-1,)`.

Once loaded, `DataViewerApp.selectDataPoint()` is called in [here](DataViewerApp), and it calls `dataPoint()` in [here](https://github.com/snu-quiqcl/iquip/blob/develop/iquip/apps/dataviewer.py#L998) with `index = (-1,)`.
And it causes an error in `data_indices = np.all(symbols.T == index, axis=1)` since `symbols.T` never has `(-1,)`.

BECATRUE, 2023-10-16:
Please check this bug! @kangz12345

kangz12345, 2023-10-16:
Great, your correction method works!

PR #204 (2023-10-17): Convert negative index to positive in `dataPoint()`
This resolves #203.

Applied the method suggested by @BECATRUE.
commit 6220dc1f (2023-10-17): Convert negative index to positive in `dataPoint()` (#204)

This resolves #203.

Applied the method suggested by @BECATRUE.
closed 2023-10-17 by kangz12345