snu-quiqcl/iquip #229: Add Scan widget

Aijuh, 2023-12-26:
### Feature you want to implement
Add widget for no scan, center scan, range scan, explicit scan.

### How the feature is implemented
1. Modify apps/builder.py line 456 to respond to scan parameters. ([scan parameter](https://github.com/m-labs/artiq/blob/master/artiq/language/scan.py))
2. Make frame for range, center, explicit scan with reference to [this](https://github.com/m-labs/artiq/blob/master/artiq/gui/entries.py).

### Additional context
Feature for transfering scan parameter to master pc will be added at the artiq-proxy repository later.

BECATRUE, 2024-03-19:
@kangz12345 I have a question in terms of UX.

In ARTIQ dashboard, an explicit scan widget offers only a string input widget and a user should write the desired sequence with delimiters. (I don't remember well, but I think it is either comma or space.)
It might seem to be crude, and a user has difficulties to see a long sequence and reorder it. But it also has an advantage that a user can copy the sequence easily.

Of course, we can make a neater UI, for example, once a space is typed, the previous string becomes an item (widget), and we can reorder them easily by dragging. (similar to hashtag in Obsidian even if they cannot be reordered.)

From the experimenter's point of view, I wonder which one is preferred, a too simple UI or a complex but functional UI.

kangz12345, 2024-03-19:
I think such features are too fancy for us as we are not going to use it that much.
Instead, supporting python script including `numpy` API, e.g., `[4, 7, 2]` or `np.linspace(0., 100., 11, endpoint=True)`, would be helpful.
Also, this can be implemented via `eval` I guess.

BECATRUE, 2024-03-21:
I see. Your suggestion looks more efficient.

@Aijuh Can you apply this decision? I think it is okay to implement it in the next issue.

If you can't afford it because of your work, I'll do it.

PR #259 (2024-03-22): Add center & explicit scan
This closes #229 

CenterScan and ExplicitScan widgets are added. Below is the image of current builder GUI. 
[image]

Center scan has `center`, `span`(length) and `step` as an argument. And Explicit scan has only one argument, `sequence`.
CenterScan widget has very similar similar to the RangeScan widget. And ExplicitScan is very simple, since it has only one argument `sequence`.

In terms of `RegExp`, I followed original code at artiq.
commit ffe022ca (2024-03-22): Add center & explicit scan (#259)

This closes #229 

CenterScan and ExplicitScan widgets are added. Below is the image of
current builder GUI.

[image]

Center scan has `center`, `span`(length) and `step` as an argument. And
Explicit scan has only one argument, `sequence`.
CenterScan widget has very similar similar to the RangeScan widget. And
ExplicitScan is very simple, since it has only one argument `sequence`.

In terms of `RegExp`, I followed original code at artiq.
closed 2024-03-22 by Aijuh