snu-quiqcl/iquip #230: Plan for updating data viewer

BECATRUE, 2023-12-28:
### Feature you want to implement
As @kangz12345 suggested, we need to update data viewer for supporting remote H5 file, dataset list viewer, table viewer, etc.

For this, I separated them into three sub-projects below:
- Explorer to show dataset list (Like artiq-dashboard data viewer)
- Table viewer to show a specific dataset (Not limited to table viewer only; may be extended further later)
- To support showing remote H5 file

### Additional context
The overall arrangement is as below.

[image]

BECATRUE, 2023-12-28:
Please check this plan! @kangz12345

kangz12345, 2023-12-29:
I think we should re-design the whole structure, instead of extend the current version.
For example, one selects a dataset, then chooses the viewer type to visualize the dataset (e.g., plot style or table) and configure it (which axes to show).

BECATRUE, 2024-01-02:
Oh, I see. Then, I will do that.