snu-quiqcl/iquip #250: Scheduler "Restart" button is not enabled

kangz12345, 2024-02-01:
### Describe the bug
AFAIK, the "Restart" button should be enabled when the scheduler fails to fetch the current schedule. (Is it right?)
However, I experienced a network disconnection (so the scheduler, dataviewer, stage controller, VSCode was disconnected), but the "Restart" button of the scheduler was not enabled.
I could see the schedule via `artiq_client show schedule` in a terminal though.
Note that the other apps were able to re-connect.

### Expected behavior
"Restart" button of the scheduler app should be enabled when it loses the connection.

### To Reproduce
I checked that this is reproduced in the following situation:
- `kangz12345/temp-stage` branch
- VPN on, using `iquip` normally
- Turn off the VPN -> all the connection lost (including scheduler, but the restart button is not enabled)
- Turn on the VPN -> Of course the button is still not enabled.

There was no error message in terminal and logger.

BECATRUE, 2024-02-02:
Thanks for your sharp observation!

I searched its detailed reason, and I realized that `websocket.recv()` cannot detect its connection loss, similar to [stackoverflow](https://stackoverflow.com/questions/25220006/web-socket-cannot-detect-client-connection-on-internet-disconnect).

To resolve it, I came up with an idea.

1. Receive a message with timeout, e.g., 5 seconds.
2. If a timeout exception occurs, to check if the connection is alive, send a ping.
3. If it returns `False`, it means that the connection is dead, thus close the thread. (It makes the restart button enabled.)

The example code is the followings.
```python
with connect(self.url) as websocket:
    while True:
        try:
            response = websocket.recv(5)
        except TimeoutError:
            if websocket.ping().wait(5):  # it means there is no change
                continue
            break                         # it means the connection is lost
        schedule = []
        ...
```

What do you think of it? @kangz12345

kangz12345, 2024-02-06:
Good to know!
I think your solution looks good enough.

PR #267 (2024-04-03): Restart fetching schedule when the connection is lost
This resolves #250.

I made sure if the "restart" button got enabled when the connection was lost due to VPN problem.