snu-quiqcl/iquip #254: Improve builder UI

kangz12345, 2024-02-29:
### Feature you want to implement
What is the feature you want to implement?

How about putting the argument, scan parameter, schedule option panels into tabs to make each builder frame smaller?

### How the feature is implemented
ex) Add *B* method to *A* class, Upgrade *C* model

### Additional context
Add any other context or screenshots about the feature here.

BECATRUE, 2024-04-03:
I've been thinking the current builder is too large, so your suggestion looks reasonable👍

Is it okay to implement this right now? @kangz12345

kangz12345, 2024-04-03:
> Is it okay to implement this right now? @kangz12345 

Yeah sure!

BECATRUE, 2024-04-04:
Using [`QTabWidget`](https://doc.qt.io/qtforpython-5/PySide2/QtWidgets/QTabWidget.html#), it seems to be implemented easily.

PR #276 (2024-04-04): Improve builder UI compactly
This closes #254.

What I have implemented is:
- Leave only the class name label, and move the experiment name in its tooltip.
- Group three list widgets; arguments, scannables, and scheduling options.
- Group two buttons.

[image]