snu-quiqcl/iquip #260: Cannot synchronize the dataset if it was already too large

kangz12345, 2024-03-22:
### Describe the bug
```
2024-03-22 17:44:31: [iquip.apps.dataviewer] Failed to synchronize the dataset.
Traceback (most recent call last):
  File "C:\Users\QC109_3\iquip\iquip\apps\dataviewer.py", line 966, in run
    self._initialize()
  File "C:\Users\QC109_3\iquip\iquip\apps\dataviewer.py", line 943, in _initialize
    rawDataset = json.loads(self.websocket.recv())
                            ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\QC109_3\.conda\envs\artiq\Lib\site-packages\websockets\sync\connection.py", line 201, in recv
    raise self.protocol.close_exc from self.recv_events_exc
websockets.exceptions.ConnectionClosedError: sent 1009 (message too big) over size limit (? > 1048576 bytes); no close frame received
```
When start sync if the dataset was already too large.
Cannot see the data.

### Expected behavior
Can see the data without any error.

BECATRUE, 2024-03-26:
Oh.. I will find how to send large dataset in chunks.

BECATRUE, 2024-03-28:
I found how to set the maximum size of websocket contents.

- [Server](https://www.uvicorn.org/#command-line-options): Use `uvicorn --ws-max-size` command. (default: 16MB)
- [Client](https://websockets.readthedocs.io/en/stable/reference/sync/client.html#opening-a-connection): Use `max_size` argument in `connect()`. (default: 1MB)

BECATRUE, 2024-03-28:
As discussed with Jiyong, I will set the maximum size for client to 2MB temporarily.

BECATRUE, 2024-03-28:
@kangz12345 How about setting it to `None`?

closed 2024-04-03 by kangz12345