snu-quiqcl/iquip #261: Missing last data

kangz12345, 2024-03-22:
### Describe the bug
When an experiment is finished, the last few data is not sent to iquip client.
For example, the last datapoint has only 132shots out of 150shots.

### Expected behavior
Last datapoint should also has entire shots.

### Expected causes and correction methods
Maybe the data arrived until the proxy is sleeping are ignored.

BECATRUE, 2024-03-25:
I also realized this problem a few days before.

In snu-quiqcl/artiq-proxy#130, to avoid the problem like this, I played a trick that it checks if there is any modifications after sleeping.
Like this, I will solve this.

kangz12345, 2024-04-03:
This is resolved, right? If it is, please close this issue.

BECATRUE, 2024-04-03:
Oh, I forgot closing this issue.
This was resolved by snu-quiqcl/artiq-proxy#132, so I close this.

closed 2024-04-03 by BECATRUE