snu-quiqcl/iquip #312: Lolenc Compatible Iquip

alexist2623, 2024-08-24:
### Feature you want to implement
To control lolenc from iquip, functionality must be added to recognize and work with both C and C++ files.

### How the feature is implemented
In `explore.py`, a code block was added to the `_addFile` method to recognize `.cpp` and `.c` files. The code ensures that these files are included in the list of files processed by the method, alongside `.py` files.

```python
    @pyqtSlot(list, object)
    def _addFile(self, experimentList: List[str], widget: Union[QTreeWidget, QTreeWidgetItem]):
        """Adds the files into the children of the widget.

        A file or directory which starts with "_" will be ignored, e.g. __pycache__/.

        Args:
            experimentList: The list of files under the widget path.
            widget: See _FileFinderThread class.
        """
        for experimentFile in experimentList:
            if experimentFile.startswith("_"):
                continue
            if experimentFile.endswith("/"):
                experimentFileItem = QTreeWidgetItem(widget)
                experimentFileItem.setText(0, experimentFile[:-1])
                # Make an empty item for indicating that it is a directory.
                QTreeWidgetItem(experimentFileItem)
            elif (
                experimentFile.endswith(".cpp") or 
                experimentFile.endswith(".c") or 
                experimentFile.endswith(".py")
            ):
                experimentFileItem = QTreeWidgetItem(widget)
                experimentFileItem.setText(0, experimentFile)

```

kangz12345, 2024-08-25:
You could use `if experimentFile.endswith((".cpp", ".c", ".py"))` instead.
See [`endswith()`](https://docs.python.org/3/library/stdtypes.html#str.endswith).

alexist2623, 2024-08-26:
Oh it can be used with tuple! thank you!